tonyc/open890 #31: Native Windows 10 build/compatibility (no WSL)

tonyc, 2021-01-30:
There's a general desire to run or build open890 on Windows 10, but without using WSL.

If anyone is interested, you can try it out by [downloading the Windows version of Elixir](https://elixir-lang.org/install.html), checking out the code, and seeing if you can get it up and running.

If that works, then it should theoretically be possible to do a binary release using Github Actions automatically.

tonyc, 2021-01-30:
Currently failing due to bakeware step failing, but if the bakeware stuff in `mix.exs` is commented out, the project builds.

```
Compiling 1 file (.ex)
Generated elixir_make app
==> bakeware
could not compile dependency :bakeware, "mix compile" failed. You can recompile this dependency with "mix deps.compile bakeware", update it with "mix deps.update bakeware" or clean it with "mix deps.clean bakeware"
==> open_890
** (Mix) "nmake" not found in the path. If you have set the MAKE environment variable,
please make sure it is correct.
```

tonyc, 2021-01-30:
Quick & dirty way to run on Windows 10 without WSL:

* Clone the repo or download the source.
* Edit `mix.exs`, and comment out/remove the following lines:

Towards the top of the file within the `def project do` block

`releases: [{:open890, release()}]` 

Towards the bottom of the file inside `def deps do`:

`{:bakeware, "~> 0.1.4"},`

* Install Elixir from here.
* Install Node.JS from here.
* Run a command prompt as administrator
* You should be able to follow the regular setup instructions from here. `npm install` from the `assets/` directory, then `mix deps.get` and `iex -S mix phx.server` from the main repo directory

tonyc, 2021-01-30:
Related issue: https://github.com/bake-bake-bake/bakeware/issues/37

tonyc, 2021-01-31:
I was able to successfully run a release that I manually built, on a separate Win10 machine that did not have Elixir/Erlang installed, so this seems quite doable.

kf40, 2021-02-01:
wonderful!  let us know how we can do this! 
Randy, KF4O

tonyc, 2021-02-03:
> wonderful! let us know how we can do this!
> Randy, KF4O

For now, you'll need to follow the manual process I outlined above. I don't have an automatic build process available for Windows, and Windows support is not at the top of my list right now (although it is not necessarily at the bottom of the list either 😉)

kf40, 2021-02-06:
Tony, sorry, when you say  "main repo directory"  do you  mean "open890-main"  directory?

tonyc, 2021-02-06:
Yep, the top-level directory of the code. You should have a `mix.exs` file in the same directory.

kf40, 2021-02-07:
okay,  all looking good now.... one last question... how do we  start the program now  (Radio on and on the web)? Sorry for all the questions.. thanks much again!

tonyc, 2021-02-07:
It's all in my comment above, and in the [README](https://github.com/tonyc/open890#config). Set your config variables, run `mix phx.server` and then browse to `http://localhost:4000`

If you still need help, we can bring it over to the discussion area.

tonyc, 2021-03-06:
Closing since I've got a path for Windows builds.

closed 2021-03-06 by tonyc