tonyc/open890 #80: Changing VFO A/B causes unexpected debug output in response to reply from radio

Tyrbiter, 2021-04-11:
Using VFO A/B switch causes the radio to reply with an unexpected FT1 response when changing to VFO B and FT0 when changing to VFO A.

This may be cosmetic, but since the VFO change is using an "FRx;" command, and gets an "FRx" response I assume there is a reason for the "FTx"

tonyc, 2021-04-11:
Thanks for the report - this is probably indeed tied to SPLIT mode, which I haven't implemented yet. I have debug messages set up for anything I am not explicitly listening for to make it easier to see what is going on - so the messages labeled `[debug]` can probably be safely ignored. :)

Tyrbiter, 2021-04-11:
I should have read through all the issues first shouldn't I?

Yes, it's a minor thing, but having the output to the terminal meant I saw it and thought it slightly odd. That's why I came back and linked it to #66

tonyc, 2021-04-11:
Ah yep. Plus you are more likely to see the debug output using a source build, compared to a released binary.

tonyc, 2021-04-11:
Whoops - I closed this prematurely. 

Thinking about this, the "Building from Source" wiki page is more of a development process, and thus, you see the `debug` messages. I will keep this issue open as a reminder to update the wiki page with a slightly different process, which will exclude the debug messages in the terminal output.

Tyrbiter, 2021-04-11:
Yes, that sounds sensible, I dived in head-first with this project with a git build and a develop checkout, one thing I was intending to do is to create a spec file so that I can build an rpm package for Fedora using a release tarball as the basis, but haven't got that far yet.

tonyc, 2021-04-11:
That would be interesting. If there is a sensible way to build the RPM using the CI process here on github, I could roll it into that process as well. The Linux builds run on Ubuntu 20.04, and I'm not sure if there are any requirements for building the RPM (if it needs to be built on a Fedora system, etc), or if an RPM can be reasonably built on an Ubuntu system and then distributed for Fedora users.