tonyc/open890 #89: Band edge frequencies are missing

tonyc, 2022-07-03:
The low/high bandscope edge frequencies seem to be missing. Where'd they go?

Tyrbiter, 2022-07-03:
I see them in hi/lo-cut mode, the display switches when I change it in menu 6-11.

tonyc, 2022-07-03:
I'm actually not sure what caused me to see this, but I'm actually getting them now that I've changed branches locally a couple times. On the radar, at least.

Tyrbiter, 2022-07-04:
I don't have Cloudlog, which may or may not be relevant.

If you don't find the cause then I can repeat your test sequence if you like and see what happens.

tonyc, 2022-07-04:
I think the band scope edges are actually working fine, but I encountered this while testing it out. So the title of this ticket might actually change.

The crash isn't related to cloudlog in any way -- it seems like the code is trying to figure out which passband "ID" to use, and something is getting mucked up since it doesn't know the filter mode. Once the mode has changed in the menu, then it all recovers and is fine so I wonder if the radio is sending the passband ID before the filter mode change.

Tyrbiter, 2022-07-04:
OK, I saw the Cloudlog debug outputs, wondered if it might affect things.

I'm going to defer to your knowledge, I just don't have enough understanding of how the various parameters are sent across and interpreted. But please shout if extra testing is needed and I can rebuild and try things pretty rapidly.

tonyc, 2022-07-04:
The cause of the problem is really annoying: the radio sends the filter info (hi/shift, low/width) BEFORE sending the message that the menu item changed, so open890 doesn't yet realize that menu 6-11 (6-12) changed. The code uses indexed lookup tables based on the command reference, so I'll need to attempt to detect this.

References #91

Tyrbiter, 2022-07-04:
I suppose the only way to deal with this is to keep the state of things that can do this so that you compare the will-change-to values it sends to the was-set-to values you already have.

But then I'm no programmer and I don't know much about web-based event-driven programming.

tonyc, 2022-07-15:
Encountered another one today while working on UI stuff. Mostly just leaving here for my own records:

```
[error] GenServer {:radio_connection_registry, {:tcp, "903e7dd4-38b7-4466-b48a-6b95b144ae44"}} terminating
** (ArithmeticError) bad argument in arithmetic expression
    :erlang.-(14101670, nil)
    (open890 0.1.0) lib/open890_web/radio_view_helpers.ex:155: Open890Web.RadioViewHelpers.offset_frequency_reverse/3
    (open890 0.1.0) lib/open890/radio_state.ex:661: Open890.RadioState.update_filter_lo_edge/1
    (open890 0.1.0) lib/open890/tcp_client.ex:307: Open890.TCPClient.handle_msg/2
    (elixir 1.12.3) lib/enum.ex:2385: Enum."-reduce/3-lists^foldl/2-0-"/3
    (open890 0.1.0) lib/open890/tcp_client.ex:76: Open890.TCPClient.handle_info/2
    (stdlib 3.14.2.2) gen_server.erl:689: :gen_server.try_dispatch/4
    (stdlib 3.14.2.2) gen_server.erl:765: :gen_server.handle_msg/6
    (stdlib 3.14.2.2) proc_lib.erl:226: :proc_lib.init_p_do_apply/3
Last message: {:tcp, #Port<0.28>, "SL025;"}
```

tonyc, 2024-07-02:
The problem here is that when the menu item changes (`EX00611 001`), the radio sends filter info down _first_, but not the reply that the menu item actually changed.

The solution may be to just "believe" that the menu item actually changed when sending the command, rather than when receiving the response from the radio.

Tyrbiter, 2024-07-02:
Presumably it doesn't send the filter info unless the menu item has changed, so you can interpret the filter info as the confirmation of the change.

Only a problem if the same thing can happen in other situations, which I don't know.

tonyc, 2024-07-02:
The problem is that you have to know what the menu setting is to properly interpret the "lo/width" and "hi/shift" messages. 

So, here's what happens when sending the command to change the menu:

- [up] change the menu item
- [down] here's the low/width
- [down] here's the high/shift
- [down] ok, the menu item changed

What _should_ happen:
- [up] change the menu item
- [down] ok, the menu item changed
- [down] here's the low/width
- [down] here's the high/shift

Anyway, I think I can get around it by just assuming the menu item changed when sending the command to change it, and then I'll know which filter mode the radio will be in, which then should cause the filter display to not crash :)