tonyc/open890 #94: Install error: install_elixir_deps Error 1

RafalLukawiecki, 2022-07-24:
Apologies if this is my error. I am trying to install open890 on macOS 12.4. I have installed adfs using brew. make up stops with error `make: *** [install_elixir_deps] Error 1` after a few successes:
```
erlang
elixir
nodejs
yarn
asdf install
elixir 1.12.3-otp-23 is already installed
erlang 23.3.4.6 is already installed
nodejs 14.15.4 is already installed
yarn 1.22.11 is already installed
mix deps.get
make: mix: No such file or directory
make: *** [install_elixir_deps] Error 1
```
Thank you for any suggestions!
73 Rafal EI6LA

tonyc, 2022-07-24:
That's odd -- after `make` installs the required asdf plugins, and all of the dependencies, you should have not only the `elixir` command available, but `mix` as well. If mix can't be found, that leads me to believe that there's something awry with your `asdf` install -- and perhaps your homebrew install as well.

What's the output you get, if you run the following commands from the checkout directory? -

```
which asdf
asdf install
elixir --version
which elixir
mix --version
which mix
```

When I had a Mac, I don't think I ever used asdf via homebrew, I always just installed it via git. If all else fails, try perhaps removing your homebrew install of asdf, and follow the "Bash & Git" instructions here https://asdf-vm.com/guide/getting-started.html#_2-download-asdf

RafalLukawiecki, 2022-07-25:
Thank you @tonyc. Neither elixir nor mix were in a path. Homebrew has not added the right paths and completions to zsh. I fixed it by adding ohmyzsh adfs extension. After rerunning make up it has finished building it. Thank you for your help!

closed 2022-07-25 by RafalLukawiecki