tonyc/open890 #95: Very small nit with notch 'drag' indicator

Tyrbiter, 2022-07-30:
When notch is enabled or disabled, the tabs for Ant.Ctrl TxRx Scope "jump" slightly and so does the waterfall.

It would be nicer if this didn't happen, I'd guess it involves a small adjustment of where the notch drag strip is placed, it probably needs to be centred on the space available.

tonyc, 2022-07-30:
You know, I saw this happen, and completely forgot about it. Tagged for 0.8.1

Tyrbiter, 2022-07-30:
> You know, I saw this happen, and completely forgot about it. Tagged for 0.8.1

If only I had remembered it earlier I could have reminded you. I didn't, mea culpa.

tonyc, 2022-10-04:
This should now be resolved - not only the jumping around, but the alignment, and I've also added a "disabled" state that changes the appearance and whether it accepts user input (click/drag/mousewheel)

closed 2022-10-04 by tonyc