tonyc/open890 #96: docker build broken?

kgarrels, 2022-07-31:
Hi,

very cool project, thanks!!

docker build is not working for me, any suggestions?

73, 
kai dm2ab

```
Last login: Sun Jul 31 09:40:37 on ttys001
kai@imac/~ > cd open890
kai@imac/~/open890 > docker build -t open890 .
[+] Building 1.9s (22/28)
 => [internal] load build definition from Dockerfile                                                                                                        0.0s
 => => transferring dockerfile: 37B                                                                                                                         0.0s
 => [internal] load .dockerignore                                                                                                                           0.0s
 => => transferring context: 34B                                                                                                                            0.0s
 => [internal] load metadata for docker.io/library/alpine:3.13                                                                                              1.0s
 => [internal] load metadata for docker.io/library/elixir:1.11.4-alpine                                                                                     1.0s
 => [stage-1 1/5] FROM docker.io/library/alpine:3.13@sha256:7769c38ce671534a0ddbf98633f17edbbd4a4e99cbc77ef42f9f7b78b6a1c6c2                                0.0s
 => [internal] load build context                                                                                                                           0.0s
 => => transferring context: 5.67kB                                                                                                                         0.0s
 => [build  1/18] FROM docker.io/library/elixir:1.11.4-alpine@sha256:92ea4b2037482e3a1e612d6aa8b4bb25512d14bb858a0cfee3a1f2be0fd995c0                       0.0s
 => => resolve docker.io/library/elixir:1.11.4-alpine@sha256:92ea4b2037482e3a1e612d6aa8b4bb25512d14bb858a0cfee3a1f2be0fd995c0                               0.0s
 => CACHED [build  2/18] RUN apk update                                                                                                                     0.0s
 => CACHED [build  3/18] RUN apk add --no-cache make gcc libc-dev nodejs yarn                                                                               0.0s
 => CACHED [build  4/18] RUN mix local.hex --force &&     mix local.rebar --force                                                                           0.0s
 => CACHED [build  5/18] RUN mkdir /app                                                                                                                     0.0s
 => CACHED [build  6/18] WORKDIR /app                                                                                                                       0.0s
 => CACHED [build  7/18] COPY config ./config                                                                                                               0.0s
 => CACHED [build  8/18] COPY lib ./lib                                                                                                                     0.0s
 => CACHED [build  9/18] COPY priv ./priv                                                                                                                   0.0s
 => CACHED [build 10/18] COPY assets ./assets                                                                                                               0.0s
 => CACHED [build 11/18] COPY mix.exs .                                                                                                                     0.0s
 => CACHED [build 12/18] COPY mix.lock .                                                                                                                    0.0s
 => CACHED [build 13/18] RUN mix deps.get                                                                                                                   0.0s
 => CACHED [build 14/18] RUN mix deps.compile                                                                                                               0.0s
 => CACHED [build 15/18] RUN yarn --cwd assets install                                                                                                      0.0s
 => ERROR [build 16/18] RUN yarn --cwd assets run deploy                                                                                                    0.7s
------
 > [build 16/18] RUN yarn --cwd assets run deploy:
#24 0.603 yarn run v1.22.19
#24 0.684 error Command "deploy" not found.
#24 0.685 info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
------
executor failed running [/bin/sh -c yarn --cwd assets run deploy]: exit code: 1
kai@imac/~/open890 >
```

tonyc, 2022-07-31:
Thanks for letting me know - I haven't touched the Docker build since changing how CSS was built, so I suspect I'll just need to update it to be more in line with how the automated release process works. I really should try to get an automated image built and pushed via Github as well one of these days.

tonyc, 2022-07-31:
Kai, please pull the latest `develop` updates, and let me know if it properly builds and starts for you now!

tonyc, 2022-07-31:
Additionally, there are two new `Makefile` targets: `make docker` and `make build_docker`. The former uses the latter, so most people using Docker should just need to run `make docker`.

closed 2022-07-31 by tonyc

kgarrels, 2022-07-31:
yes, works perfect now, and pulling the ready-made docker image is even more elegant.

Thanks and best 73,
kai

tonyc, 2022-07-31:
Great! The only thing I can see that is a problem is that you lose the connection and config information when the container is stopped and restarted - this is because it uses the `db/` and `config/` directories on the filesystem to store the database containing the connection info.

I've just committed a `docker-compose.yml` file to the `develop` branch that appropriately maps the volumes, so you can also use `docker-compose up` to start the prebuilt image with the volumes mapped. This will allow the connection db and configs to live across restarts.

kgarrels, 2022-08-02:
now we are entering the realm of luxury :-)