tonyc/open890 #97: Waterfall filter indicator incorrect when in FSK mode

RafalLukawiecki, 2022-08-25:
I have noticed that the filter indicator seems to be away from the centre frequency when operating in FSK mode. This does NOT always happen, usually when changing bands or modes. It is as if the calculation for the location of the filter switched between using the MARK frequency and the frequency of MARK with the usual LSB offset. See screenshot below. The Tx and Rx point to the correct frequency, but the filter marker is off by over 2 kHz lower down.

[image]

tonyc, 2022-08-25:
Thanks for the report!

RafalLukawiecki, 2022-12-17:
Perhaps this may help: the issue happens when switching bands. The fix is to change mode on the radio from FSK to CW and then back to FSK. At that point the width of the filter passband returns to where it should be.

tonyc, 2022-12-23:
@RafalLukawiecki I believe I've managed to track this one down - thank you for the clue! It is committed to `main` and `develop`, and will be included in the next release as well.

closed 2022-12-23 by tonyc

RafalLukawiecki, 2022-12-23:
Many thanks, @tonyc. I will reply once I had a chance to test it again.

commit 3532de19 (2022-12-23): Don't shift the passband polygon in FSK/PSK modes. Resolves #97