tonyc/open890 #98: Release 1.0.0 - notch drag bar not centred

Tyrbiter, 2022-11-05:
As for the title, the notch drag bar seems to be left justified.

tonyc, 2022-11-05:
Thanks! I thought I had it centered at one point, but it must have reverted. I'll get it fixed up before 1.0.0 is actually released.

Tyrbiter, 2022-11-05:
Just thought I'd mention it, yes it was fine but I suppose it's down to changes in the Tx/Ant tab affecting that whole area.

tonyc, 2022-11-06:
@Tyrbiter I've made some UI tweaks around the slider, I've re-centered the notch indicator and added some border-radius so they look a little nicer. The changes are on `develop` and `main` branches, I plan on re-cutting the 1.0.0-pre release today.

[image]

closed 2022-11-06 by tonyc

Tyrbiter, 2022-11-06:
Had a quick look and it is now fixed, quite a nice look overall.