tonyc/open890 #99: Analogue meter

tonyc, 2022-11-18:
Implement the 890's analogue meter. Allow switching between analogue/digital

Tyrbiter, 2022-11-20:
Had a quick look at the WIP, so far so good.

I suspect you may have fun trying to match the needle dynamics but then Kenwood did it so it is possible ;-)

tonyc, 2022-11-21:
Yeah, although my guess is they are reading the actual analog meter values, i.e. what get sent through the meter port on the back. The radio only sends digital data between from 0 to 70, so that's why it looks a bit choppy.

[image]

I _am_, however, looking into simulating the needle physics 😎  - I found this useful article while researching it all https://hackaday.io/project/185180-rack-mount-midi-meters/log/206251-simulating-needle-physics

I may also build a "digilog" (i.e. digital-analog) meter that replicates how the TS-590's meter looks, by using pips arranged in an arc:

[image]

Tyrbiter, 2022-11-21:
Well it looks fun, and I suppose the choppiness can be dealt with by some interpolation. An analogue meter mechanism is essentially spring/mass/damper from the vibrations and waves module taught in an EE degree, but then I did all that 40 years ago and no doubt I'm rusty.