tonyc/open890 #102: Bandscope grid - possible?

Tyrbiter, 2022-11-24:
It would be useful to have some frequency and amplitude divisions on the bandscope area of the display as shown on the TS890.

I'd guess this is fairly simple to implement, but it's not really a missing feature, just nice to have.

tonyc, 2022-11-24:
Hm, I'm not quite sure what you mean - does the grid not show up for you?

[image]

Tyrbiter, 2022-11-24:
Clearly not, I had not really noticed before but currently with the whole of 20m across the bandscope I have no grid lines.

Firefox 107 on Fedora 37.

Strange.

tonyc, 2022-11-24:
Interesting - I wonder if this is a local thing just to you, related to CSS -- IIRC we were dealing with some strangeness with NodeJS on your system at one point. I wonder if you've got a cached CSS file somewhere perhaps, from an old build.

At the risk of making things not work for you, the main thing I would suggest is a `make clean` followed by `make up`. You could also try a fresh checkout somewhere, to maybe rule out the caching thing.

Tyrbiter, 2022-11-24:
OK, I have created a new checkout of the code and run make up.

I recreated the connection to the radio from scratch.

I don't see any grid in the bandscope.

Sorry that I seem to have my special break everything system in use ;-)

tonyc, 2022-11-25:
Ahh ok, I think you found a missing feature! I think I never implemented the frequency grid for fixed mode. I will change the title of this issue to reflect that.

Tyrbiter, 2022-11-25:
Glad that is sorted out, not quite sure how all the grid lines disappeared initially but I suppose that was probably a local effect.

tonyc, 2022-11-27:
In fixed mode, vertical grids appear at the following intervals for the spans specified:

| span (kHz) | interval (kHz) |
| -----------|---------|
| 5 - 9 | 0.5 |
| 10 - 19 | 1 |
| 20 - 29 | 2 |
| 30 - 49 | 3 |
| 50 - 99 | 5 |
| 100 - 199 | 10 |
| 200 - 499 | 20 |
| 500 | 50 |

tonyc, 2022-12-01:
@Tyrbiter This is available on `develop` now. Please check it out and let me know if you find anything weird related to this

closed 2022-12-01 by tonyc