tonyc/open890 #103: It won't compile under Mint Vanessa.

ON5HB, 2022-11-25:
Hi Tony,

I run Mint Vanessa, based on Ubuntu Jammy.
Sorry to say but it won't compile from source, all sorts of stuff is missing.
Also your binaries don't work either, complain about missing libraries.
And tried the docker way, it refuses also.

Too bad, as I really wanted to test it.

Ps. got stuff like this:

bas@workstation:~/open890$ make
mix clean --deps
rm -rf deps/
rm -rf priv/static_assets
rm -rf assets/node_modules
/bin/sh: 1: asdf: not found
/bin/sh: 1: asdf: not found
make: *** [Makefile:21: install_asdf_plugins] Fout 127
bas@workstation:~/open890$ 
bas@workstation:~$ docker pull ghcr.io/tonyc/open890:latest
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/images/create?fromImage=ghcr.io%2Ftonyc%2Fopen890&tag=latest": dial unix /var/run/docker.sock: connect: permission denied
bas@workstation:~$ sudo docker pull ghcr.io/tonyc/open890:latest
Error response from daemon: received unexpected HTTP status: 503 Service Unavailable
bas@workstation:~$ 

And I do not like docker, rather run the binary.

Bas - ON5HB.

tonyc, 2022-11-25:
Hi Bas,

Thank you for the report. Based on the output you pasted, it looks as if you don't have `asdf` installed. Please see the instructions at https://github.com/tonyc/open890/wiki/Installing-From-Source for more information, as asdf is the recommended way to build open890 from source, and the makefile assumes asdf is installed.

Would you mind sharing output from when you try to run the Linux binary? It may give us some hints as to which libraries are absolutely required for a binary install to run.

I have recently switched from Ubuntu to Mint for my Linux boot, although I haven't built open890 from source on that machine yet.

ON5HB, 2022-11-25:
I tried to install ASDF, but it's not a part of Mint. I find so many ASDF stuff, I do not know what to install.
You may want to compile it yourself to find the issues. 

Thanks Bas.

tonyc, 2022-11-25:
Hi Bas,

If asdf is not in the Mint repos, you should be able to install it directly via git: 

https://asdf-vm.com/guide/getting-started.html#_2-download-asdf

Then follow step 3 for the shell you are using.

tonyc, 2022-11-25:
Hi Bas,

Unfortunately we will have to disagree on this matter - I don't have any interest in including every possible dependency in this way.

closed 2022-11-25 by tonyc