tonyc/open890 #105: Audio scope filter shape disappears

Tyrbiter, 2022-12-05:
I have noticed a few instances of the disappearing filter shape and I think the LC and HC labels, the values are still shown

This gives long strings of this in the console window:

[debug] Unimplemented case for audio_scope_filter_edges for mode nil

It gives the impression of something being lost, possibly on swapping VFOs or maybe switching from V to M or the reverse.

Refreshing the browser window fixes it, is it maybe a caching issue?

tonyc, 2022-12-06:
It seems like it's not properly retrieving the current mode, so it doesn't know how to draw the scope edges. If you can reliably reproduce it to figure out what causes it, I'd greatly appreciate it!

Tyrbiter, 2022-12-06:
It seems to happen when I switch to Memory mode from VFO mode, it's consistent when I switch back and forth but I can get the filter shape to appear again by refreshing the browser tab and from then on switching VFO and Memory mode doesn't seem to lose it. It would seem to be something that doesn't initialise on server start, I can stop the server, restart and essentially the same sequence occurs.

tonyc, 2022-12-08:
Thanks! That should help quite a bit when I get around to looking at this.

Tyrbiter, 2022-12-26:
The same issue also happens when Disconnecting and then Reconnecting. Swapping VFOs fixes it immediately.

tonyc, 2022-12-30:
@Tyrbiter I think I've tracked this one down, and pushed a couple of fixes to `develop` - mind checking it out?

Tyrbiter, 2022-12-30:
Well it starts, but as soon as I open the browser tab for Radio UI I get errors, here is the full output from starting the server:

mix phx.server
[debug] RadioConnectionRepo.init()
[debug] db/ exists, skipping
[info] Running Open890Web.Endpoint with cowboy 2.9.0 at 0.0.0.0:4000 (http)
[info] Access Open890Web.Endpoint at http://localhost:4000
[info] *** Cloudlog Supervisor started
[info] *** UDP audio server: start_link
[info] *** UDP audio server: init
[info] UDP Audio server listening on port 60001
                       ___ ___  ___
   ___  ___  ___ ___  ( _ ) _ \/ _ \
  / _ \/ _ \/ -_) _ \/ _  \_, / // /
  \___/ .__/\__/_//_/\___/___/\___/
     /_/

  open890 is now running.
  Access the web interface at http://localhost:4000

[info] Auto-starting connection id fa8467e1-de2d-4a33-9923-49c72ea79123, "TS890"
[info] Cloudlog not enabled for connection fa8467e1-de2d-4a33-9923-49c72ea79123
[info] Established TCP socket with radio on port 60000
[info] [UP] "##CN;"
[watch] build finished, watching for changes...
[info] [UP] "##ID10410bdm2Squrrox667;"
[info] signed in, scheduling first ping
[info] Enabling audio scope via LAN
[info] [UP] "DD11;"
[info] Enabling LAN bandscope
[info] [UP] "DD01;"
[info] [UP] "AI2;"
[info] [UP] "FR;"
[info] [DN] "##TI1"
[debug] RadioState.dispatch: unhandled message: "##TI1"
[info] [UP] "FR;"
[info] [UP] "FA;"
[info] [UP] "FB;"
[info] [UP] "BU0;"
[info] [UP] "BU1;"
[info] [UP] "SM;"
[info] [UP] "EX00611;"
[info] [UP] "EX00612;"
[info] [UP] "MV;"
[info] [UP] "OM0;"
[info] [UP] "OM1;"
[info] [UP] "SH0;"
[info] [UP] "SL0;"
[info] [UP] "FS;"
[info] [UP] "FL0;"
[info] [UP] "FL10;"
[info] [UP] "FL11;"
[info] [UP] "FL12;"
[info] [UP] "BSO;"
[info] [UP] "BS3;"
[info] [UP] "BS4;"
[info] [UP] "BSM0;"
[info] [UP] "BSA;"
[info] [UP] "BS8;"
[info] [UP] "BS5;"
[info] [UP] "BY;"
[info] [UP] "DS1;"
[info] [UP] "PA;"
[info] [UP] "RA;"
[info] [UP] "BSC;"
[info] [UP] "MA70;"
[info] [UP] "MA71;"
[info] [UP] "RM11;"
[info] [UP] "RM21;"
[info] [UP] "RM51;"
[... 155 lines cut ...]
[error] #PID<0.573.0> running Open890Web.Endpoint (connection #PID<0.553.0>, stream id 3) terminated
Server: localhost:4000 (http)
Request: GET /connections/0c9a8a63-8714-4a37-a7ed-dee7b3bba5cd?panel=true&panelTab=txrx
** (exit) an exception was raised:
    ** (KeyError) key :id not found in: nil. If you are using the dot syntax, such as map.field, make sure the left-hand side of the dot is a map
        (open890 0.1.0) lib/open890_web/components/buttons.ex:624: anonymous fn/2 in Open890Web.Components.Buttons.pop_out_bandscope_button/1
        (phoenix_live_view 0.17.5) lib/phoenix_live_view/diff.ex:387: Phoenix.LiveView.Diff.traverse/7
        (phoenix_live_view 0.17.5) lib/phoenix_live_view/diff.ex:486: anonymous fn/4 in Phoenix.LiveView.Diff.traverse_dynamic/7
        (elixir 1.14.2) lib/enum.ex:2468: Enum."-reduce/3-lists^foldl/2-0-"/3
        (phoenix_live_view 0.17.5) lib/phoenix_live_view/diff.ex:387: Phoenix.LiveView.Diff.traverse/7
        (phoenix_live_view 0.17.5) lib/phoenix_live_view/diff.ex:486: anonymous fn/4 in Phoenix.LiveView.Diff.traverse_dynamic/7
        (elixir 1.14.2) lib/enum.ex:2468: Enum."-reduce/3-lists^foldl/2-0-"/3
        (phoenix_live_view 0.17.5) lib/phoenix_live_view/diff.ex:387: Phoenix.LiveView.Diff.traverse/7
        (phoenix_live_view 0.17.5) lib/phoenix_live_view/diff.ex:486: anonymous fn/4 in Phoenix.LiveView.Diff.traverse_dynamic/7
        (elixir 1.14.2) lib/enum.ex:2468: Enum."-reduce/3-lists^foldl/2-0-"/3
        (phoenix_live_view 0.17.5) lib/phoenix_live_view/diff.ex:387: Phoenix.LiveView.Diff.traverse/7
        (phoenix_live_view 0.17.5) lib/phoenix_live_view/diff.ex:138: Phoenix.LiveView.Diff.render/3
        (phoenix_live_view 0.17.5) lib/phoenix_live_view/static.ex:234: Phoenix.LiveView.Static.to_rendered_content_tag/4
        (phoenix_live_view 0.17.5) lib/phoenix_live_view/static.ex:126: Phoenix.LiveView.Static.render/3
        (phoenix_live_view 0.17.5) lib/phoenix_live_view/controller.ex:38: Phoenix.LiveView.Controller.live_render/3
        (phoenix 1.6.15) lib/phoenix/router.ex:354: Phoenix.Router.__call__/2
        (open890 0.1.0) lib/open890_web/endpoint.ex:1: Open890Web.Endpoint.plug_builder_call/2
        (open890 0.1.0) lib/plug/debugger.ex:136: Open890Web.Endpoint."call (overridable 3)"/2
        (open890 0.1.0) lib/open890_web/endpoint.ex:1: Open890Web.Endpoint.call/2
        (phoenix 1.6.15) lib/phoenix/endpoint/cowboy2_handler.ex:54: Phoenix.Endpoint.Cowboy2Handler.init/4

tonyc, 2022-12-30:
Hmm, this shouldn't have caused this error. It looks like maybe you got the connection id in the URL bar incorrect somehow, maybe accidentally deleted a character or something? 

Does it work when you navigate back to `/connections` in your browser and click the "Radio" button?

Tyrbiter, 2022-12-30:
Right, restarted server and then closed Radio UI tab, then reloaded connections page and reopened Radio UI. Working again, some sort of browser caching thing?

I now get the filter shape when stopping and restarting and when changing VFOs, however the filter shape is gone when I switch from V->M and back when I go from M->V.

HTH.

tonyc, 2022-12-30:
Great, I think I'm on the disappearing filter, then. I'll add a little bit of logic on the radio page, that if for some reason it can't find the requested connection ID, it will just redirect you back to the connection index screen, which should prevent the big nasty error you saw.

Tyrbiter, 2022-12-30:
Sounds good, I can see from looking at the commands that some clever tricks are needed to keep track of filter settings.

Really appreciate the effort to fix this up.

tonyc, 2022-12-30:
Could you tell me specifically what your VFO frequency/modes are, and likewise for the memory channel you're switching between? And whether either the VFO or memory channel are in SPLIT?

I saw the disappearing filter briefly, but then managed to stop seeing it.

Tyrbiter, 2022-12-30:
VFOA is 20m USB, VFOB is 80m LSB, Mem channel is 60m USB

tonyc, 2022-12-30:
Great, thanks!

tonyc, 2023-04-02:
@Tyrbiter After a bit of a hiatus, I'm picking this back up. I've pushed a couple commits to develop that should help. I wasn't able to completely replicate the filter disappearing, but I did notice that the filter was displaying the "wrong" sideband in a couple cases.

If you've got time and are able, would you mind checking out latest `develop` and see if that helps the issue?

Tyrbiter, 2023-04-02:
OK, I have just made a quick test, I only got time to do this just before I head for bed :)

I still see the filter shape vanish as soon as I select Memory instead of VFO. Both LC and HC also vanish but the :<freq> that follows them is still there.

Here is a complete dump of the console output, please let me know if you want any more information:

[bdm@deangelis open890-develop]$ make up
(asdf plugin-list | grep erlang && asdf plugin-update erlang) || asdf plugin-add erlang
erlang
Updating erlang...
Already on 'master'
Your branch is up to date with 'origin/master'.
(asdf plugin-list | grep elixir && asdf plugin-update elixir) || asdf plugin-add elixir
elixir
Updating elixir...
Already on 'master'
Your branch is up to date with 'origin/master'.
(asdf plugin-list | grep nodejs && asdf plugin-update nodejs) || asdf plugin-add nodejs
nodejs
Updating nodejs...
Already on 'master'
Your branch is up to date with 'origin/master'.
(asdf plugin-list | grep yarn && asdf plugin-update yarn) || asdf plugin-add yarn
yarn
Updating yarn...
Already on 'master'
KERL_CONFIGURE_OPTIONS="--without-javac --without-fop --without-odbc" asdf install
elixir 1.14.2-otp-24 is already installed
erlang 24.3.4 is already installed
nodejs 18.12.1 is already installed
yarn 1.22.11 is already installed
mix deps.get
Resolving Hex dependencies...
Resolution completed in 0.074s
Unchanged:
  castore 0.1.13
  certifi 2.8.0
  cowboy 2.9.0
  cowboy_telemetry 0.3.1
  cowlib 2.11.0
  esbuild 0.4.0
  file_system 0.2.10
  floki 0.32.0
  gettext 0.18.2
  hackney 1.18.0
  html_entities 0.5.2
  httpoison 1.8.0
  idna 6.1.1
  jason 1.4.0
  metrics 1.0.1
  mime 2.0.3
  mimerl 1.2.0
  mix_test_watch 1.1.0
  parse_trans 3.3.1
  phoenix 1.6.15
  phoenix_html 3.2.0
  phoenix_live_reload 1.3.3
  phoenix_live_view 0.17.5
  phoenix_pubsub 2.1.1
  phoenix_template 1.0.0
  phoenix_view 2.0.2
  plug 1.14.0
  plug_cowboy 2.6.0
  plug_crypto 1.2.3
  ranch 1.8.0
  ssl_verify_fun 1.1.6
  telemetry 0.4.3
  telemetry_metrics 0.6.1
  telemetry_poller 0.5.1
  toml 0.6.2
[... 461 lines cut ...]
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "FR0"
[info] [DN] "FT0"
[debug] RadioState.dispatch: unhandled message: "FT0"
[info] [DN] "MV0"
new vfo_memory_state: :vfo
[info] [DN] "RA0"
[info] [DN] "BS80"
[info] [DN] "BSC040"
[info] [DN] "BSM01400000014350000"
[info] [DN] "BSM11400000014350000"
[debug] RadioState.dispatch: unhandled message: "BSM11400000014350000"
[info] [DN] "BSM21400000014100000"
[debug] RadioState.dispatch: unhandled message: "BSM21400000014100000"
[info] [DN] "BSM31410000014350000"
[debug] RadioState.dispatch: unhandled message: "BSM31410000014350000"
[info] [DN] "BU01"
[info] [DN] "BU12"
[info] [DN] "DF0000000000000"
[debug] RadioState.dispatch: unhandled message: "DF0000000000000"
[debug] Unable to find cloudlog process for connection id fa8467e1-de2d-4a33-9923-49c72ea79123
[debug] Bandscope LV: RX connection_state: :stopped

Tyrbiter, 2023-04-02:
And of course no sooner do I send the previous message and after I restarted the connection in the browser, now it doesn't happen and the filter shape stays put.

More console output which follows on from above:

[debug] Bandscope LV: RX connection_state: :starting
[info] Established TCP socket with radio on port 60000
[info] [UP] "##CN;"
[debug] Bandscope LV: RX connection_state: :up
[info] [UP] "##ID10410bdm2Squrrox667;"
[info] signed in, scheduling first ping
[info] Enabling audio scope via LAN
[info] [UP] "DD11;"
[info] Enabling LAN bandscope
[info] [UP] "DD01;"
[info] [UP] "AI2;"
[info] [UP] "FR;"
[debug] *** GET INITIAL STATE ***
[info] [DN] "##TI1"
[debug] RadioState.dispatch: unhandled message: "##TI1"
[warning] Unknown vfo_memory_state: nil
[warning] Unknown vfo_memory_state: nil
[warning] Unknown vfo_memory_state: nil
[warning] Unknown vfo_memory_state: nil
[warning] Unknown vfo_memory_state: nil
[warning] Unknown vfo_memory_state: nil
[warning] Unknown vfo_memory_state: nil
[warning] Unknown vfo_memory_state: nil
[info] [UP] "FR;"
[info] [UP] "FA;"
[info] [UP] "FB;"
[info] [UP] "BU0;"
[info] [UP] "BU1;"
[info] [UP] "SM;"
[info] [UP] "EX00611;"
[info] [UP] "EX00612;"
[info] [UP] "MV;"
[warning] Unknown vfo_memory_state: nil
[info] [UP] "OM0;"
[warning] Unknown vfo_memory_state: nil
[info] [UP] "OM1;"
[info] [UP] "SH0;"
[info] [UP] "SL0;"
[info] [UP] "FS;"
[info] [UP] "FL0;"
[info] [UP] "FL10;"
[info] [UP] "FL11;"
[info] [UP] "FL12;"
[info] [UP] "BSO;"
[info] [UP] "BS3;"
[info] [UP] "BS4;"
[info] [UP] "BSM0;"
[info] [UP] "BSA;"
[info] [UP] "BS8;"
[info] [UP] "BS5;"
[info] [UP] "BY;"
[info] [UP] "DS1;"
[info] [UP] "PA;"
[info] [UP] "RA;"
[info] [UP] "BSC;"
[info] [UP] "MA70;"
[info] [UP] "MA71;"
[info] [UP] "RM11;"
[info] [UP] "RM21;"
[info] [UP] "RM51;"
[info] [UP] "DD0;"
[info] [UP] "AG;"
[info] [UP] "RG;"
[info] [UP] "PC;"
[info] [UP] "NT;"
[info] [UP] "NW;"
[... 780 lines cut ...]
[debug] RadioState.dispatch: unhandled message: "FT1"
[info] [DN] "MV0"
new vfo_memory_state: :vfo
[info] [DN] "OM11"
[info] [DN] "PA2"
[info] [DN] "RA0"
[info] [DN] "SH1022"
[debug] RadioState.dispatch: unhandled message: "SH1022"
[info] [DN] "SL103"
[debug] RadioState.dispatch: unhandled message: "SL103"
[info] [DN] "BS80"
[info] [DN] "BSC040"
[info] [DN] "BSM02800000028500000"
[info] [DN] "BSM12800000028500000"
[debug] RadioState.dispatch: unhandled message: "BSM12800000028500000"
[info] [DN] "BSM22800000028200000"
[debug] RadioState.dispatch: unhandled message: "BSM22800000028200000"
[info] [DN] "BSM32820000028700000"
[debug] RadioState.dispatch: unhandled message: "BSM32820000028700000"
[info] [DN] "BU01"
[info] [DN] "BU12"
[info] [DN] "DF0000000000000"
[debug] RadioState.dispatch: unhandled message: "DF0000000000000"
[debug] Unable to find cloudlog process for connection id fa8467e1-de2d-4a33-9923-49c72ea79123
[debug] Bandscope LV: RX connection_state: :stopped

tonyc, 2023-04-03:
Yeah, this is a weird one. In your first log, it looks like the radio is not sending the operating mode (the `OM` command) when you switch to memory mode, and so it doesn't know how to render the filter edges in the audio scope:

```
[info] [DN] "MV1"
new vfo_memory_state: :memory
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "RA1"
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "BS80"
[info] [DN] "BSC040"
[info] [DN] "BSM00500000005500000"
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "BSM10500000005500000"
[debug] RadioState.dispatch: unhandled message: "BSM10500000005500000"
[info] [DN] "BSM20500000005500000"
[debug] RadioState.dispatch: unhandled message: "BSM20500000005500000"
[info] [DN] "BSM30500000005500000"
[debug] RadioState.dispatch: unhandled message: "BSM30500000005500000"
[info] [DN] "BU00"
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "BU10"
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "DF0000000000000"
[debug] RadioState.dispatch: unhandled message: "DF0000000000000"
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "FR0"
[info] [DN] "FT0"
[debug] RadioState.dispatch: unhandled message: "FT0"
[info] [DN] "MV0"
new vfo_memory_state: :vfo
```

Here it looks like you are switching from VFO to memory, and then back to VFO mode. But in this span, the radio doesn't send the operating mode, which is really annoying.

But in your second log, it does:

```
new vfo_memory_state: :memory
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "OM02"
```

That last `OM02` command is "left side USB"

I was hoping to squash this one for the next release, I'll keep messing with it a bit tomorrow, but if I can't reliably get it to work, it may just linger.

Tyrbiter, 2023-04-03:
I'm not expecting miracles, but it's clear that the radio responses do seem to vary depending on how it's setup and what span is set for each VFO. My test involved pressing only the M/V button.

I'm going to assume that all TS-890s behave the exactly same way, but if you need me to test and send more debug please ask, it would be embarrassing for Kenwood to have to admit that the firmware is non-deterministic :)

To avoid any doubt I have fw 1.04

Tyrbiter, 2023-04-05:
Just noticed this:

If TS-890 is in VFO mode when I connect in the browser tab then switching to Memory mode loses the filter passband shape and LC/HC annotations.

If I disconnect while in Memory mode, then reconnect, the filter passband shape appears together with the LC and HC annotations but then switching to VFO mode (and swapping VFOs) loses them shape and annotations again.

Here is a complete debug dump of this after running make up to get to the develop branch immediately after 1.0.4 in main:

mix phx.server
[debug] RadioConnectionRepo.init()
[debug] db/ exists, skipping
[info] Running Open890Web.Endpoint with cowboy 2.9.0 at 0.0.0.0:4000 (http)
[info] Access Open890Web.Endpoint at http://localhost:4000
[info] *** Cloudlog Supervisor started
[info] *** UDP audio server: start_link
[info] *** UDP audio server: init
[info] UDP Audio server listening on port 60001
                       ___ ___  ___
   ___  ___  ___ ___  ( _ ) _ \/ _ \
  / _ \/ _ \/ -_) _ \/ _  \_, / // /
  \___/ .__/\__/_//_/\___/___/\___/
     /_/

  open890 is now running.
  Access the web interface at http://localhost:4000

[info] Auto-starting connection id fa8467e1-de2d-4a33-9923-49c72ea79123, "TS890"
[info] Cloudlog not enabled for connection fa8467e1-de2d-4a33-9923-49c72ea79123
[info] Established TCP socket with radio on port 60000
[info] [UP] "##CN;"
[watch] build finished, watching for changes...
[info] [UP] "##ID10410bdm2Squrrox667;"
[info] signed in, scheduling first ping
[info] Enabling audio scope via LAN
[info] [UP] "DD11;"
[info] Enabling LAN bandscope
[info] [UP] "DD01;"
[info] [UP] "AI2;"
[info] [UP] "FR;"
[debug] *** GET INITIAL STATE ***
[info] [DN] "##TI1"
[debug] RadioState.dispatch: unhandled message: "##TI1"
[info] [UP] "FR;"
[info] [UP] "FA;"
[info] [UP] "FB;"
[info] [UP] "BU0;"
[info] [UP] "BU1;"
[info] [UP] "SM;"
[info] [UP] "EX00611;"
[info] [UP] "EX00612;"
[info] [UP] "MV;"
[info] [UP] "OM0;"
[info] [UP] "OM1;"
[info] [UP] "SH0;"
[info] [UP] "SL0;"
[info] [UP] "FS;"
[info] [UP] "FL0;"
[info] [UP] "FL10;"
[info] [UP] "FL11;"
[info] [UP] "FL12;"
[info] [UP] "BSO;"
[info] [UP] "BS3;"
[info] [UP] "BS4;"
[info] [UP] "BSM0;"
[info] [UP] "BSA;"
[info] [UP] "BS8;"
[info] [UP] "BS5;"
[info] [UP] "BY;"
[info] [UP] "DS1;"
[info] [UP] "PA;"
[... 1759 lines cut ...]
[info] [DN] "RA0"
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "BS80"
[info] [DN] "BSC040"
[info] [DN] "BSM01400000014350000"
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "BSM11400000014350000"
[debug] RadioState.dispatch: unhandled message: "BSM11400000014350000"
[info] [DN] "BSM21400000014100000"
[debug] RadioState.dispatch: unhandled message: "BSM21400000014100000"
[info] [DN] "BSM31410000014350000"
[debug] RadioState.dispatch: unhandled message: "BSM31410000014350000"
[info] [DN] "BU01"
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "BU12"
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[info] [DN] "DF0000000000000"
[debug] RadioState.dispatch: unhandled message: "DF0000000000000"
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[debug] Unimplemented case for audio_scope_filter_edges for mode nil
[debug] Unable to find cloudlog process for connection id fa8467e1-de2d-4a33-9923-49c72ea79123
[debug] Bandscope LV: RX connection_state: :stopped

tonyc, 2023-04-06:
Argh, memory mode!!!!

[image]

This thing is such a pain in the rear! :) On the radar for a 1.0.5 cleanup.

tonyc, 2023-04-06:
Note for myself: I think the solution is to send `OM` to retrieve the left/right operating modes when receiving and `MV` response and the active mode is `nil`

closed 2024-06-30 by tonyc

Tyrbiter, 2024-06-30:
That seems to work, not too much code changed too :)

tonyc, 2024-06-30:
Woohoo!

commit 6d4f4f22 (2024-06-30): Re-query the active mode when toggling M/V. Probably closes #105