tonyc/open890 #106: Browser tab recognition - add a recognisable icon

Tyrbiter, 2022-12-07:
Could a nice colourful icon be added so that the browser in use shows an easily to spot tab?

I'm an inveterate tab user and finding the right one is a lot easier with a visual cue.

This could also stand as a project badge on github.

tonyc, 2022-12-08:
Something like this?

[image]

Not exactly "colorful" I guess - but it would fit the Kenwood aesthetic :)

There are some other tricks I could do, like dynamically changing the icon based on e.g. mode changes. So it could show "890" for a few seconds and then cycle to "USB", and then back.

Tyrbiter, 2022-12-08:
That's certainly possible. I was wondering about the little icon that appears on the radio when KNS is active as the basis for this, but I'm not in a position to insist :)

Whatever you think appropriate, you're the one who has done all the work :)

commit 37efe53d (2022-12-23): Add favicons. Closes #106
closed 2022-12-23 by tonyc