tonyc/open890 #116: User-defined markers

tonyc, 2023-05-08:
Have the ability to create/place a marker on the spectrum scope, through keyboard and also through a UI button click.

Also have the ability to remove a specific marker, or clear all markers at once.

The markers should be associated to the connection, and persist across restarts.

tonyc, 2023-05-17:
This has landed in develop, with a couple of caveats:

* Markers are not yet persisted across page reloads or connection restarts
* Keyboard controls only (no UI yet) - `mm`, `mr` / `mg` / `mb` to drop white/red/green/blue markers, and `cc` / `cr` / `cg` / `cb` to clear white/red/green/blue markers.

tonyc, 2023-05-17:
Also to note, you don't necessarily need to have the mouse cursor hovering over the spectrum - the keyboard shortcuts will currently drop the marker at the left-side VFO location, not where the mouse is pointing.

Tyrbiter, 2023-05-17:
Yes, I had that selected, although I don't remember doing it so maybe it was a default I've never changed.

So if the cursor is over the spectrum it should drop the marker where the cursor is or have you not implemented that yet?

In any case I now get markers, I will wait for development to get it all sorted out.

tonyc, 2023-05-17:
> So if the cursor is over the spectrum it should drop the marker where the cursor is or have you not implemented that yet?

I haven't decided on this interaction yet, but it seems like a reasonable way to expect it to behave!

Tyrbiter, 2023-05-17:
It looks like being a useful feature. Perhaps you could also have a list of fixed markers that are always present, a bit like the macros in config.toml and user configurable. Marking frequencies for unusual modes is the application I'm thinking of, like the freedv frequencies for instance.