tonyc/open890 #120: Yarn Install Error

joshuajayg, 2023-07-28:
I'm trying to build on a RPi4 and came across an error in the build while running `make up`.

asdf is installed. 

Here's a dump of my log:
```
:~/open890 $ make up
(asdf plugin-list | grep erlang && asdf plugin-update erlang) || asdf plugin-add erlang
erlang
Updating erlang to master
Already on 'master'
Your branch is up to date with 'origin/master'.
(asdf plugin-list | grep elixir && asdf plugin-update elixir) || asdf plugin-add elixir
elixir
Updating elixir to master
Already on 'master'
Your branch is up to date with 'origin/master'.
(asdf plugin-list | grep nodejs && asdf plugin-update nodejs) || asdf plugin-add nodejs
nodejs
Updating nodejs to master
Already on 'master'
Your branch is up to date with 'origin/master'.
(asdf plugin-list | grep yarn && asdf plugin-update yarn) || asdf plugin-add yarn
yarn
Updating yarn to main
Already on 'main'
Your branch is up to date with 'origin/main'.
KERL_CONFIGURE_OPTIONS="--without-javac --without-fop --without-odbc" asdf install
elixir 1.14.2-otp-24 is already installed
erlang 24.3.4 is already installed
nodejs 18.12.1 is already installed
yarn 1.22.11 is already installed
mix deps.get
warning: the VM is running with native name encoding of latin1 which may cause Elixir to malfunction as it expects utf8. Please ensure your locale is set to UTF-8 (which can be verified by running "locale" in your shell)
Resolving Hex dependencies...
Resolution completed in 0.828s
Unchanged:
  castore 0.1.13
  certifi 2.8.0
  cowboy 2.9.0
  cowboy_telemetry 0.3.1
  cowlib 2.11.0
  esbuild 0.4.0
  file_system 0.2.10
  floki 0.32.0
  gettext 0.18.2
  hackney 1.18.0
  html_entities 0.5.2
  httpoison 1.8.0
  idna 6.1.1
  jason 1.4.0
  metrics 1.0.1
  mime 2.0.3
  mimerl 1.2.0
  mix_test_watch 1.1.0
  parse_trans 3.3.1
  phoenix 1.6.15
  phoenix_html 3.2.0
  phoenix_live_reload 1.3.3
  phoenix_live_view 0.17.5
  phoenix_pubsub 2.1.1
  phoenix_template 1.0.0
  phoenix_view 2.0.2
  plug 1.14.0
  plug_cowboy 2.6.0
  plug_crypto 1.2.3
  ranch 1.8.0
  ssl_verify_fun 1.1.6
  telemetry 0.4.3
  telemetry_metrics 0.6.1
  telemetry_poller 0.5.1
  toml 0.6.2
  unicode_util_compat 0.7.0
  uniq 0.5.3
All dependencies are up to date
mix deps.compile
warning: the VM is running with native name encoding of latin1 which may cause Elixir to malfunction as it expects utf8. Please ensure your locale is set to UTF-8 (which can be verified by running "locale" in your shell)
===> Analyzing applications...
===> Compiling parse_trans
===> Analyzing applications...
===> Compiling mimerl
===> Analyzing applications...
===> Compiling metrics
===> Analyzing applications...
===> Compiling unicode_util_compat
===> Analyzing applications...
===> Compiling idna
===> Analyzing applications...
===> Compiling ranch
===> Analyzing applications...
===> Compiling telemetry
===> Analyzing applications...
===> Compiling telemetry_poller
===> Analyzing applications...
===> Compiling certifi
===> Analyzing applications...
===> Compiling hackney
===> Analyzing applications...
===> Compiling cowlib
===> Analyzing applications...
===> Compiling cowboy
===> Analyzing applications...
===> Compiling cowboy_telemetry
mix compile
warning: the VM is running with native name encoding of latin1 which may cause Elixir to malfunction as it expects utf8. Please ensure your locale is set to UTF-8 (which can be verified by running "locale" in your shell)
Generated open890 app
yarn install --cwd assets
/home/pi/.asdf/lib/commands/command-exec.bash: line 28: /home/pi/.asdf/installs/nodejs/18.12.1/bin/node: No such file or directory
make: *** [Makefile:38: yarn_install] Error 127
```

tonyc, 2023-07-31:
Thanks for the report! At a surface level, it looks like `node` isn't in your PATH for some reason. I just acquired a new RPi4 myself, so I can try to take a look at this and see if I can replicate it on my own.

What do you see when you run `node --version` ?

joshuajayg, 2023-07-31:
I think I had open890 installed on this Pi before but it actually might have been a Pi3, I don't remember.  

Here's the result of `node --version`.  Looks like it isn't configured correctly.
```
No version is set for command node
Consider adding one of the following versions in your config file at 
nodejs 18.12.1
```

tonyc, 2023-07-31:
OK, that's the problem then - you don't have node in your path. I recommend using `asdf` to manage development dependencies - it should be as simple as `asdf install` once you've got it installed to get everything. 

If you don't want to install asdf, then you'll need to find a way to get node 18 installed.

joshuajayg, 2023-07-31:
Interesting, I went through the `asdf` installation.  Maybe I missed something in the installation and it didn't go right.  I'll try today when I get home.

joshuajayg, 2023-08-03:
I checked `asdf` and it wasn't showing it was installed though I had the PATH set and the files were there so I removed them, ran `git clone . . . ` again, and now `asdf` is working right.  Yarn installed this time around.

joshuajayg, 2023-08-12:
I jumped the gun on closing this.  I reinstalled asdf but I'm still getting an error.  Here's the log again.  

```
pi@KJ7LVZ-Home:~/open890 $ make up
(asdf plugin-list | grep erlang && asdf plugin-update erlang) || asdf plugin-add erlang
erlang
Updating erlang to master
Already on 'master'
Your branch is up to date with 'origin/master'.
(asdf plugin-list | grep elixir && asdf plugin-update elixir) || asdf plugin-add elixir
elixir
Updating elixir to master
Already on 'master'
Your branch is up to date with 'origin/master'.
(asdf plugin-list | grep nodejs && asdf plugin-update nodejs) || asdf plugin-add nodejs
nodejs
Updating nodejs to master
Already on 'master'
Your branch is up to date with 'origin/master'.
(asdf plugin-list | grep yarn && asdf plugin-update yarn) || asdf plugin-add yarn
yarn
Updating yarn to main
Already on 'main'
Your branch is up to date with 'origin/main'.
KERL_CONFIGURE_OPTIONS="--without-javac --without-fop --without-odbc" asdf install
elixir 1.14.2-otp-24 is already installed
erlang 24.3.4 is already installed
nodejs 18.12.1 is already installed
yarn 1.22.11 is already installed
mix deps.get
warning: the VM is running with native name encoding of latin1 which may cause Elixir to malfunction as it expects utf8. Please ensure your locale is set to UTF-8 (which can be verified by running "locale" in your shell)
Resolving Hex dependencies...
Resolution completed in 0.873s
Unchanged:
  castore 0.1.13
  certifi 2.8.0
  cowboy 2.9.0
  cowboy_telemetry 0.3.1
  cowlib 2.11.0
  esbuild 0.4.0
  file_system 0.2.10
  floki 0.32.0
  gettext 0.18.2
  hackney 1.18.0
  html_entities 0.5.2
  httpoison 1.8.0
  idna 6.1.1
  jason 1.4.0
  metrics 1.0.1
  mime 2.0.3
  mimerl 1.2.0
  mix_test_watch 1.1.0
  parse_trans 3.3.1
  phoenix 1.7.7
  phoenix_html 3.3.1
  phoenix_live_reload 1.3.3
  phoenix_live_view 0.19.5
  phoenix_pubsub 2.1.1
  phoenix_template 1.0.0
  phoenix_view 2.0.2
  plug 1.14.0
  plug_cowboy 2.6.0
  plug_crypto 1.2.3
  ranch 1.8.0
  ssl_verify_fun 1.1.6
  telemetry 0.4.3
  telemetry_metrics 0.6.1
  telemetry_poller 0.5.1
  toml 0.6.2
  unicode_util_compat 0.7.0
[... 170 lines cut ...]
warning: undefined attribute "padded_top" for component Open890Web.Components.Slider.slider/1
  lib/open890_web/live/radio.html.heex:227: (file)

warning: undefined attribute "padded_top" for component Open890Web.Components.Slider.slider/1
  lib/open890_web/live/radio.html.heex:232: (file)

warning: undefined attribute "padded_top" for component Open890Web.Components.Slider.slider/1
  lib/open890_web/live/radio.html.heex:237: (file)

Generated open890 app
yarn install --cwd assets
/home/pi/.asdf/lib/commands/command-exec.bash: line 28: /home/pi/.asdf/installs/nodejs/18.12.1/bin/node: No such file or directory
make: *** [Makefile:38: yarn_install] Error 127
```
I checked to see if node existed.
```
pi@KJ7LVZ-Home:~/open890 $ ls /home/pi/.asdf/installs/nodejs/18.12.1/bin/
corepack  node  npm  npx
```
```
pi@KJ7LVZ-Home:~/open890 $ node --version
/home/pi/.asdf/lib/commands/command-exec.bash: line 28: /home/pi/.asdf/installs/nodejs/18.12.1/bin/node: No such file or directory
```

What's going on here?

tonyc, 2023-08-12:
Huh - honestly it still looked like a borked asdf install to me:

```bash
tonyc@Tony-Windows-PC ~/d/open890 (develop)> node --version
v18.12.1
tonyc@Tony-Windows-PC ~/d/open890 (develop)> which node
/home/tonyc/.asdf/shims/node
tonyc@Tony-Windows-PC ~/d/open890 (develop)> cat ~/.asdf/shims/node
#!/usr/bin/env bash
# asdf-plugin: nodejs 18.12.1
exec /home/tonyc/.asdf/bin/asdf exec "node" "$@" # asdf_allow: ' asdf '
tonyc@Tony-Windows-PC ~/d/open890 (develop)> ~/.asdf/bin/asdf exec node
Welcome to Node.js v18.12.1.
Type ".help" for more information.
>
tonyc@Tony-Windows-PC ~/d/open890 (develop)> ~/.asdf/bin/asdf exec node --version
v18.12.1
```

I'm curious if `asdf reshim nodejs` alleviates anything?

joshuajayg, 2023-08-14:
No, `asdf reshim nodejs` did not help.  
`asdf install nodejs` comes back with 
`nodejs 18.12.1 is already installed`

joshuajayg, 2023-08-14:
If I type `node` from my home directory, I get 
```No version is set for command node
Consider adding one of the following versions in your config file at 
nodejs 18.12.1
```
but if I go into the open890 directory, I get the error that the file does not exist.

tonyc, 2023-08-17:
weird, it kinda seems like a messed-up asdf install. The only other thing I can think of is to have you completely remove asdf from your system and try a reinstall

joshuajayg, 2024-02-09:
I reinstalled OSX on my Macbook for security reasons and tired open890 again.  Same problem.  I think it was an error on my part adding asdf to my PATH.  There is a not after installing asdf with Homebrew that says you need to add a line to your terminal profile and it gives you a few options for locations.  When listing the current files in my home directory, none of those files existed.  Rather than following their directions directly, I added `. /usr/local/opt/asdf/libexec/asdf.sh` to `~/.zprofile`.  `make up` compiled fine after that.

closed 2024-02-09 by joshuajayg