tonyc/open890 #124: RF Power Slider, nothing happens

lb5va, 2023-08-12:
Why can it be that I'm not able to move/contyrol this new feature, the RF Power slide?
(Release 1.0.6)
Also no modulation. Any hints of controlling the modulation with a client computer?  
(I have sound in Windows with default mic.)

tonyc, 2023-08-13:
Hi Stig - thank you for the report!

I already have an issue open for the RF power slider in #122 and is already fixed in the `develop` branch of the source code. I expect a release in the next day or two to resolve it.

The ability to transmit audio via open890 is tracked in #6, but I have no ETA.

closed 2023-08-13 by tonyc

lb5va, 2023-08-13:
OK, so what modes are the RF-powertglider for then...
#LB5VA_Stig_______73_ 
Vennlig hilsen Stig Rasmussen LB5VA Tlf.    2227 1687 Mob. 4004 1687 

    På søndag

tonyc, 2023-08-13:
The power slider works for any mode, exactly like the POWER knob on the physical radio

lb5va, 2023-08-13:
Thanks!
I really like your remote system for the TS890, and hoping the SSB TX audio also will work, as I almost only run SSB.
The scope is really much better than on the radio, Ref. and Scale adjustments. 

Thanks also for answering my questions/remarks so fast.

#LB5VA_Stig_______73_ 

    På søndag