tonyc/open890 #130: Open890 1.0.8.1 Docker problem

OH6OKSA, 2024-01-08:
Hi,

I have been running Open890 on virtualbox virtual machine and it has been working fine. But when I updated to latest 1.0.8.1 version this issue occured:

12:51:12.631 [error] Could not check origin for Phoenix.Socket transport.

 

Origin of the request: http://192.168.1.138:4000

 

This happens when you are attempting a socket connection to

a different host than the one configured in your config/

files. For example, in development the host is configured

to "localhost" but you may be trying to access it from

"127.0.0.1". To fix this issue, you may either:

 

  1. update [url: [host: ...]] to your actual host in the

     config file for your current environment (recommended)

 

  2. pass the :check_origin option when configuring your

     endpoint or when configuring the transport in your

     UserSocket module, explicitly outlining which origins

     are allowed:

 

        check_origin: ["https://example.com",

                       "//another.com:888", "//other.com"]

I can access the web page and all configuration can be done. But there is no output from radio to webinterface. 

Also tried with virtual host ip...same problem:
localhost:~# docker run --env OPEN_HOST=192.168.1.138 -p 4000:4000 -p 60001:6000
1/udp -it --rm ghcr.io/tonyc/open890:latest
13:27:37.888 [info] Running Open890Web.Endpoint with cowboy 2.9.0 at :::4000 (http)
13:27:37.894 [info] Access Open890Web.Endpoint at http://localhost:4000
13:27:37.897 [info] *** Cloudlog Supervisor started
13:27:37.900 [info] 

*** UDP audio server: start_link

13:27:37.900 [info] 

*** UDP audio server: init

13:27:37.900 [info] UDP Audio server listening on port 60001
                       ___ ___  ___
   ___  ___  ___ ___  ( _ ) _ \/ _ \
  / _ \/ _ \/ -_) _ \/ _  \_, / // /
  \___/ .__/\__/_//_/\___/___/\___/
     /_/

  open890 is now running.  Press ^C^C (ctrl-c, ctrl-c) to stop.

  Access the web interface at http://localhost:4000/

  You can change the hostname and port by setting OPEN890_HOST
  and OPEN890_PORT environment variables respectively.

tonyc, 2024-01-08:
Hi there!

It looks like you might have used a slightly incorrect ENV var. You wrote:

`OPEN_HOST=192.168.1.138`

The correct var is `OPEN890_HOST`

You may also have a typo in your port forwarding. You had: `60001:6000`, but you probably just want to use `60001:60001`

Give that a shot and tell me if it resolves your problem!

OH6OKSA, 2024-01-08:
THX, sorry was my bad...some copy paste mistake :-(

THX for your Great development work!

closed 2024-01-08 by tonyc