tonyc/open890 #132: Is it possible change audio UDP port for another

bv3un, 2024-04-06:
Dear Sir .is it possible to change audio UDP port from 60001 to another or can customize ?
because some ISP are ban UDP over 60000 .if it can customize on open890 then we can do the redirection port forward from outside to 890 on our router .for example 
                                                                                             home side
outside open 890 udp 50055 ------internet ---------modem--(router port forward 50055/60001) ---TS 890

thanks

tonyc, 2024-04-06:
I'll have to make the UDP audio port configurable, but it's entirely possible. I'll add it to the list. Thanks for the request!

tonyc, 2024-04-06:
@bv3un I've just added the ability to configure the UDP port via the ENV var `OPEN890_UDP_PORT` - it works just like `OPEN890_HOST` and `OPEN890_PORT`. This is now in `main` and will be in the next binary release as well.

If you're able to do a dev build, please test it and let me know if any more changes need to be made to it. Thanks!

closed 2024-04-06 by tonyc

bv3un, 2024-04-14:
Hi Sir 
thanks .but i am not familiar with code .
  
i just  modify runtime.exs under and modify UDP port from 60001 to which i like to ? for example 33500 and make up again ? 

  udp_port = System.get_env("OPEN890_UDP_PORT", "**33500**") |> String.to_integer()
  udp_port = Enum.min([udp_port, 65535]) 

and what is Enum.min stand for ?

thanks

tonyc, 2024-04-14:
Hi @bv3un,

You do not need to change any code, my recent changes allow you to set it from the outside while starting open890, e.g. on Linux:

`OPEN890_UDP_PORT=33350 ./bin/open890.sh`

Or in Windows:

```
set OPEN890_UDP_PORT=33500
bin\open890.bat
```

If you're unable to do a local development build to get the latest changes, please be patient while I finish a few other features, and the ability to set the port will be in the next binary release :)

btw Enum.min returns the minimum value out of the list, e.g.:

```elixir
Enum.min([50, 1, 4]) # => returns 1
```

Cheers